PhysicsUofRAUI/binTempSensor #22: Edit cable design

PhysicsUofRAUI, 2021-02-07:
Need to place filter capcitors most likely and perhaps change assembly methods. 

Use this method to find the right design (at the start)
1. One sensor at the end with of a 30 m wire with the capacitor near it. 
- if it does not work put a capacitor near the arduino too
- if that does not work remove the capacitor close to the sensor.
2. Add the other two sensors incrementally (change up capacitor positions somewhat logically).
3. Once it is working create a circuit diagram.

khanafin, 2021-03-24:
Interesting Project...  I have developed a similar system for collecting temps underwater in lakes.  I use an ESP32 to send the data from TI tmp-107 sensors to a server via wifi or cellular 2G/3G/LTE.  I have 15-20 sensors monitoring the thermocline 150m offshore.  Sample data:
[image]
Let me know if you are interested in talking.  
Karl Hanafin
karl@hanafins.com

PhysicsUofRAUI, 2021-07-30:
Using the DS18B20 sensors for the most part now so this is not needed anymore

closed 2021-07-30 by PhysicsUofRAUI